raspberrypi/rpi-eeprom #700: https boot with secure boot seems impossible

27o, 2025-04-29:
### Describe the bug

https booting fails with a programmed boot key.
http booting after removing HTTP_CACERT_HASH works fine.

The only difference in the boot process apart from the hashes and signatures is ```5.90 Unsupported boot order 7```

HTTP_CACERT_HASH is the correct hash as it worked without customer key in OTP and is the same as sha256sum of the extracted cacert.der and its original input.

### Steps to reproduce the behaviour

Burn a secure boot key to OTP.
Try booting using https

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

```ini
[all]
BOOT_UART=1
BOOT_ORDER=0xf7
HTTP_CACERT_HASH=96bcec06264976f37460779acf28c5a7cfe8a3c0aae11a8ffcee05c0bddf08c6
HTTP_HOST=boot.example.com
HTTP_PATH=net_install
```

### Bootloader logs

```
  1.52 RPi: BOOTSYS release VERSION:897f94ed DATE: 2025/03/27 TIME: 00:16:44
  1.56 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1743034604 serial 341702b4 boardrev c04170 stc 1856267
  1.66 AON_RESET: 00000003 PM_RSTS 00001020
  1.70 POWER_OFF_ON_HALT: 0 WAIT_FOR_POWER_BUTTON 0 power-on-reset 0
  1.80 RP1_BOOT chip ID: 0x20001927
  1.84 PCIEx1: PWR 0 DET_WAKE 0
  1.84 part 00000000 reset_info 00000000
  1.88 PMIC reset-event 00000000 rtc 6810aa4c alarm 00000000 enabled 0
  1.94 uSD voltage 3.3V
  2.13 Initialising SDRAM rank 2 total-size: 32 Gbit 4267 (0x15 0x00)
  2.16 DDR 4267 1 0 32 152 BL:1
  3.53 OTP boardrev c04170 bootrom a a
  3.55 Customer key hash de8083...
  3.62 VC-JTAG unlocked
  3.85 RP1_BOOT chip ID: 0x20001927
  3.90 bootconf.sig
  3.90 hash: c332f3...
  3.96 rsa2048: c066ae....
  4.43 RSA verify
  4.59 rsa-verify pass (0x0)

  4.87 RP1_BOOT chip ID: 0x20001927
  4.87 RP1_BOOT: fw size 46888
  5.25 PCI2 init
  5.25 PCI2 reset
  5.70 PCIe scan 00001de4:00000001
  5.70 RP1_CHIP_INFO 20001927

  5.73 RPi: BOOTLOADER release VERSION:897f94ed DATE: 2025/03/27 TIME: 00:16:44
  5.80 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1743034604 serial 341702b4 boardrev c04170 stc 5380428
  5.90 Unsupported boot order 7
  5.15 usb_pd_init status 3
  5.15 USB_PD CONFIG 0 41
  5.17 Boot mode: RESTART (0f) order 0
  5.21 Restart 0 max -1
  5.23 Boot mode: RESTART (0f) order 0
  5.26 Restart 0 max -1
  5.28 Boot mode: RESTART (0f) order 0
  5.32 Restart 0 max -1
  5.34 Boot mode: RESTART (0f) order 0
```

tdewey-rpi, 2025-04-29:
Thanks for the report, @27o 

This isn't what I would have expected to happen, so I'll take a look.

peterharperuk, 2025-04-29:
I think it is mistakenly disabling HTTP if HTTP_CACERT_HASH is specified rather than disabling it if HTTP_HOST is NOT specified. This is probably an issue from when we first implemented network install. I could probably send you an test bootloader to try if you are prepared to give it a go?

27o, 2025-04-29:
Sure, i flashed it more than a few times figuring out what went wrong.

peterharperuk, 2025-05-06:
I forgot to attach a test image. See attached from which you can extract the pieeprom.bin file.

[rpi-eeprom-recovery.zip](https://github.com/user-attachments/files/20061270/rpi-eeprom-recovery.zip)

27o, 2025-05-06:
This one booted fine.

Only thing i found was a warning in the kernel and invalid data in the device tree /base/reserved-memory/nvram@0/reg (raspberrypi,bootloader-config)
```
[    0.000000] OF: reserved mem: OVERLAP DETECTED!
[    0.000000] atf@0 (0x0000000000000000--0x0000000000080000) overlaps with nvram@0 (0x0000000000038000--0x00000000000382c3)
```

This only happens on the secure boot one and not on a pristine one. Both are using the new pieeprom.bin
Reading the bootloader config is impossible this way but the data for the public key (nvram@1, raspberrypi,bootloader-public-key) is fine.

Xamarin2019, 2025-06-05:
Sorry for the offtopic, but I too have encountered a problem when enabling secure boot mode. My system image after boot obviously works differently than before going to secure boot and I haven't figured it out yet, but I see the message too: 
```
[    0.000000] OF: reserved mem: OVERLAP DETECTED!
[    0.000000] atf@0 (0x0000000000000000--0x0000000000080000) overlaps with nvram@0 (0x0000000000038000--0x00000000000382c3)
```
What could it be caused by? How can it be fixed?