raspberrypi/rpi-eeprom #701: CM5 secure boot mass storage gadget requires *unsigned* bootcode

karalabe, 2025-04-29:
### Describe the bug

I have an issue open against https://github.com/raspberrypi/usbboot/issues/304

I was trying to secure boot an RPI CM5 and launch the mass storage gadget. After failing to do so for 3 hours, I figured out that the **signed bootcode is rejected** and if I replace it with an **unsigned bootcode**, that is accepted, even though the boot.img later is checked for the signature.

Seems that there is a fault and the CM5 does **NOT** properly enforce secure boot.

Do note, I did not burn my secure boot key yet into the board, rather ran with `SIGNED_BOOT=1`

### Steps to reproduce the behaviour

Enable secure boot on a CM5 via usbboot:

```
cd secure-boot-recovery5
echo `SIGNED_BOOT=1` >> boot.conf
../tools/update-pieeprom.sh -f -k $KEY
../rpiboot -d .
```

Sign the mass storage gadget:

```
cd mass-storage-gadget64
./sign.sh $KEY
../rpiboot -d .
```

^ the above will fail to boot:

```
Sending bootcode.bin
Failed control transfer (-7,24)
Failed to write correct length, returned -7
Waiting for BCM2835/6/7/2711/2712...
```

Nuke out the signature from the bootcode, by reverting it to the **original, unsigned**:

```
rm bootfiles.bin
mv bootfiles.original.bin bootfiles.bin
../rpiboot -d .
```

Boom, MSG boots, with an unsigned bootcode:

```
% ../rpiboot -d .
RPIBOOT: build-date 2025/04/29 pkg-version local ecfb7222

Please fit the EMMC_DISABLE / nRPIBOOT jumper before connecting the power and USB cables to the target device.
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.

Loading: ./bootfiles.bin
Using ./bootfiles.bin
Waiting for BCM2835/6/7/2711/2712...

Sending bootcode.bin
Successful read 4 bytes 
Waiting for BCM2835/6/7/2711/2712...

Second stage boot server
File read: mcb.bin
File read: memsys00.bin
File read: memsys01.bin
File read: memsys02.bin
File read: memsys03.bin
File read: bootmain
Loading: ./boot.sig
File read: boot.sig
Loading: ./boot.img
File read: boot.img
Second stage boot server done
```

### Device (s)

Raspberry Pi CM5

### Bootloader configuration.

Please see above

tdewey-rpi, 2025-04-30:
> I was trying to secure boot an RPI CM5 and launch the mass storage gadget. After failing to do so for 3 hours, I figured out that the signed bootcode is rejected and if I replace it with an unsigned [...]

I fear the docs could be made clearer, as it appears you've misunderstood the requirement for signed boot.

Signed boot is not *enforced* until you write the customer key hash into OTP, using the program_pubkey flag with a recovery bootloader. Additionally, you can only enable the additional security features (Development key revocation, JTAG lock-out) when the OTP key has been written.

Including a boot.sig is a useful testing step - to show you're forming the signature file correctly. It's not intended for production.

Why does the signed variant fail? Because it doesn't have an OTP key to compare to.

As such, I'm closing this as 'by design', but I will explore if we can make the docs clearer. FWIW, I cannot strongly recommend reading the source of rpi-sb-provisioner if you're curious about the requirements of the process.

closed 2025-04-30 by tdewey-rpi

karalabe, 2025-04-30:
Ah, makes sense. I guess it mostly boils down to docs being a bit on the skimpy side wrt secure boot.

Reading the provisioner does help, I do it quite a bit, but I didn't want to burn the OTP just yet. However, without doing that, the official usbboot docs don't seem to work. I kind of understand what the issue is, just would possibly say it's unexpected since it would be nice to have a full "enf-to-end" secure boot test before burning the key.

Either way, no biggie, opened these issues mostly for others hitting the same to find. 

Thanks

pelwell, 2025-04-30:
> FWIW, I cannot strongly recommend reading the source of rpi-sb-provisioner

Huh?

tdewey-rpi, 2025-04-30:
> > FWIW, I cannot strongly recommend reading the source of rpi-sb-provisioner
> 
> Huh?

Too many comments on-the-go.

tdewey-rpi, 2025-04-30:
> Ah, makes sense. I guess it mostly boils down to docs being a bit on the skimpy side wrt secure boot.
>
> Reading the provisioner does help, I do it quite a bit, but I didn't want to burn the OTP just [...]

Understood.

FWIW, you can probably achieve this through a multi-stage flow:

1) Use rpi-sb-provisioner in the fde-only mode to set up a LUKS container, but _not_ signed boot.
2) Use the SIGNED_BOOT only test-path to confirm your end-to-end flow
3) Once you're confident, re-run rpi-sb-provisioner, but this time in secure-boot mode.