raspberrypi/rpi-eeprom #702: Memory timing might be too tight

bukatlib, 2025-05-05:
### Describe the bug

I have noticed that a prime sieve might give me an incorrect count of primes mostly shortly after the boot.

I do not think that it is a SW bug since sanitizers report no errors and I observed both:

1) the incorrect count (some ones were zeroed)
2) the correct count but later write of the memory image to file contained some bits cleared (cleared between the counting and write to a binary file)

There is one example for the second case:
```
$ cmp -l memory_primes_right.img memory_primes_right_differ.img
5985247185 100   0
5985247187  40   0
5985247189   2   0
5985247191  21   0
5985247192  40   0
5985247193 100   0
5985247194 100   0
5985247195  42   0
5985247196   1   0
5985247197   1   0
5985247201  44   0
5985247202   4   0
5985247204   2   0
5985247209  10   0
5985247210 202   0
5985247211  54   0
5985247212  20   0
5985247213 200   0
5985247214 100   0
5985247215   2   0
5985247216 200   0
```

It seems to me that memory refresh interval is too long which maybe depends on the temperature or other tuning parameters. Note that memory images have 6750 MB (primes to 100 G) and I have not observed these mismatched on x86_64 architecture.

### Steps to reproduce the behaviour

I may provide the source code if desired, or other memory tester writing patterns and reading them with a delay (under the full load) may help.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

[all]
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf146
SDRAM_BANKLOW=1

### System

Raspberry Pi reference 2024-07-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 48efb5fc5485fafdc9de8ad481eb5c09e1182656, stage5

vcgencmd bootloader_version
2025/03/27 00:16:44
version 897f94edff080e2b325930669fa4e9caf469e090 (release)
timestamp 1743034604
update-time 1746369352
capabilities 0x0000007f

vcgencmd version
2025/03/27 00:16:44 
Copyright (c) 2012 Broadcom
version 897f94ed (release) (embedded)

uname -a
Linux raspberrypi 6.12.25-bpf+ #1 SMP PREEMPT Sun May  4 15:25:58 CEST 2025 aarch64 GNU/Linux

stu-spp, 2025-05-05:
Yes, the source code is desired. I would like to try to reproduce your results.

bukatlib, 2025-05-06:
Here is the code for Eratosthenes sieve:
[EratosthenesSieve.zip](https://github.com/user-attachments/files/20052829/EratosthenesSieve.zip)

Steps to build it:
```
$ mkdir build; cd build
$ cmake ..
$ make -j4
```

To count primes to 100G and write 6.75 GB memory image to disk (hardcoded to memory_primes.img for testing purposes), run the following command (close memory hungry apps beforehand):
```
$ ./prime_sieve_bench 100000000000 -
sieve size: 49999999999 bits (6250000000 bytes)
Sieve initialized in 2.25054 secs
Primes sieved in 15.8042 secs.
Found 4118054813 primes in 0.507386 secs (bandwidth 12318 MB/s).
Binary image written in 20.9711 secs.
```

I expect the following sha256 hash of the memory `footprint` is (openssl instead of sha256sum since it is much faster):
```
$ openssl dgst -sha256 memory_primes_right.img
SHA2-256(memory_primes_right.img)= bc773de13417112ba5df5bd234069ea489528d3c0d4cfdc4bf8e35d438adb323
```

Note that the incorrect results are occasionally observed shortly after the boot.

timg236, 2025-05-06:
It seems unlikely that it's memory timing related. Try boosting the core-voltage e.g over_voltage_delta=40000 in-case it's cache voltage margin issue on this board.

popcornmix, 2025-05-06:
Also, do you have any cooling on the Pi5?
Your test code does seem like quite a hard stress test, and temperature throttles quickly on an uncooled board.

bukatlib, 2025-05-06:
There is a simple script I created for a stress test:
```
#!/bin/bash

for i in $(seq 10); do
    ./prime_sieve_bench 100000000000 -
    sha256=$(openssl dgst -sha256 memory_primes.img | awk '{print $2}')
    if [ "${sha256}" != "bc773de13417112ba5df5bd234069ea489528d3c0d4cfdc4bf8e35d438adb323" ]; then
        mv memory_primes.img memory_primes_invalid_${i}.img
        echo "Invalid ${sha256} sum of sieve image." 1>&2
    else
        rm memory_primes.img
    fi
done

exit 0
```

I use an active cooler, there are the sensors values under the load:
```
CPU frequencies:
2400 2400 2400 2400 

load average:
2,32, 1,60, 1,26

rp1_adc-isa-0000
Adapter: ISA adapter
in1:           1.47 V  
in2:           2.53 V  
in3:           1.37 V  
in4:           1.42 V  
temp1:        +54.3°C  

pwmfan-isa-0000
Adapter: ISA adapter
fan1:        6027 RPM

cpu_thermal-virtual-0
Adapter: Virtual device
temp1:        +62.8°C  

rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A  

nvme-pci-10100
Adapter: PCI adapter
Composite:    +42.9°C  (low  = -273.1°C, high = +89.8°C)
                       (crit = +94.8°C)
Sensor 1:     +42.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +40.9°C  (low  = -273.1°C, high = +65261.8°C
```

pelwell, 2025-05-06:
Could you run the memtester at the same time as the prime sieve?

bukatlib, 2025-05-06:
It is interesting that I reproduced it again in the first iteration of 10 after the boot but not the others:
```
sieve size: 49999999999 bits (6250000000 bytes)
Sieve initialized in 1.09111 secs
Primes sieved in 14.8186 secs.
Found 4118054813 primes in 0.495501 secs (bandwidth 12613.5 MB/s).
Binary image written in 20.6925 secs.
Invalid c5ae25e8b8e2b920c1b91d544828a3e90c8a6a661c5f00919f4522cf0667859f sum of sieve image.
sieve size: 49999999999 bits (6250000000 bytes)
Sieve initialized in 2.22051 secs
Primes sieved in 15.5266 secs.
Found 4118054813 primes in 0.500797 secs (bandwidth 12480.1 MB/s).
Binary image written in 20.2859 secs
...
```

```
$ cmp -l memory_primes_invalid_1.img memory_primes_right.img
2889326545   0 200
2889326547   0 300
2889326549   0  40
2889326550   0 101
2889326551   0 244
2889326552   0   4
2889326553   0   2
2889326557   0   4
2889326558   0 100
2889326560   0   2
2889326564   0  10
2889326565   0 100
2889326566   0   4
2889326568   0  20
2889326569   0  60
2889326571   0   1
2889326572   0  40
2889326574   0   2
```

bukatlib, 2025-05-06:
Note that I have not overclocked rpi5 so far. What is the safe `over_voltage_delta` and is there something else what should be set, e.g., `force_turbo`? Are there some config options to disable some memory optimizations in EEPROM?

bukatlib, 2025-05-06:
Started the sieve stress load and memtester in parallel on a reduce set size (50 G sieve, 3500M memtest) and observed no errors at least in the first memtest loop. Tomorrow I will try again.

bukatlib, 2025-05-07:
Note that all the cleared bytes in the provided examples are within 64-byte range:
```
(gdb) print 2889326574>>6
$1 = 45145727
(gdb) print 2889326545>>6
$2 = 45145727
(gdb) print 5985247216>>6
$3 = 93519487
(gdb) print 5985247185>>6
$4 = 93519487
```

If I am not wrong, the cache line size is 64 bytes, it might be related.

timg236, 2025-05-07:
> Note that I have not overclocked rpi5 so far. What is the safe `over_voltage_delta` and is there something else what should be set, e.g., `force_turbo`? Are there some config options to disable some memory optimizations in EEPROM?

The firmware and hardware won't allow unsafe voltages but it will get warmer. 40mV should be plenty but you could try up to 100mV

bukatlib, 2025-05-07:
Thank you for the information, tried +20 mV, +40 mV, and +50 mV and still able to reproduce. Is someone else able to reproduce it? I start to think that I have a defective piece...

bukatlib, 2025-05-08:
Under clocked to 2 GHz and still managed to reproduce it. It is easiest to reproduce when RPI 5 starts cold (after power-off and from CPU temperature), when the system "heats up", it works more reliably. Tested again on x86_64 with primes to 1000 G and no errors observed.

timg236, 2025-05-08:
Any reason for using SDRAM_BANKLOW=1? I've set a test running on a Pi5 16GB, no failures yet. If you can post the serial number then I can check manufacturing logs in-case anything stands out (unlikely)

bukatlib, 2025-05-08:
There is a serial number:
770d9ba7ecbc92f7

The reason why I added SDRAM_BANKLOW=1 is that it is recommended for Pi8 8GB with NUMA support. I will try to collect `vclog -m` logs with and without SDRAM_BANKLOW set and post it when available. Thank you for the help.

timg236, 2025-05-08:
I've checked the manufacturing test logs and the test results show a normal first-time pass with typical voltage ranges. I've left the test running in a soak tests and will try again under-voltage to see if that can provoke an error on a board which seems to pass with default values.

popcornmix, 2025-05-08:
> The reason why I added SDRAM_BANKLOW=1 is that it is recommended for Pi8 8GB with NUMA support. I will try to collect vclog -m logs with and without SDRAM_BANKLOW set and post it when available. Thank you for the help.

I suspect the results will be the same. If `SDRAM_BANKLOW` is not specified, it will default to `1` on a Pi5, so best to remove it, and let the default setting do it's job.

bukatlib, 2025-05-08:
I managed to provoke the error again (requires a few attempts + reboots):
```
sieve size: 49999999999 bits (6250000000 bytes)
Sieve initialized in 1.12518 secs
Primes sieved in 15.2058 secs.
Found 4118054792 primes in 0.501467 secs (bandwidth 12463.4 MB/s).
Binary image written in 19.8811 secs.
Invalid b475e825943edd386c8a601f9bf7af92242007983fbd1889556754b69f632fc2 sum of sieve image.
```

There is the output of `vclog -m`:
```
006029.132: Initial voltage 800000 temp 35082
006229.725: avs_2712: AVS pred 8410 841000 temp 35082
006233.333: vpred 841 mV +0
006861.675: FB framebuffer_swap 1
006881.604: Select resolution HDMI0/2 hotplug 1 max_mode 2
006898.029: HDMI0 edid block 0 offset 0
006900.409: 00ffffffffffff000469a32401010101
006906.081: 1d18010380342078ea4ca5a7554da226
006911.754: 105054230800818081409500a940b300
006917.427: d1c001010101283c80a070b023403020
006923.100: 360006442100001a000000fd00323d1e
006928.773: 5311000a202020202020000000fc0050
006934.445: 423234380a20202020202020000000ff
006940.118: 0045374c4d51533032373033360a0121
006958.166: HDMI0 edid block 1 offset 128
006960.723: 02031ef14b900504030201111213141f
006966.396: 230907078301000065030c0010001a36
006972.069: 80a070381e403020350006442100001a
006977.742: 662156aa51001e30468f330006442100
006983.415: 001e011d007251d01e206e2855000644
006989.087: 2100001e8c0ad08a20e02d10103e9600
006994.760: 064421000018011d8018711c1620582c
007000.433: 250006442100009e00000000000000fe
007006.124: HDMI0: best-mode 2 (limit 2) 1920x1080 60 Hz CEA modes 3e001f80000000000000000000000000 extensions 1
007017.718: Select resolution HDMI1/2 hotplug 0 max_mode 2
007024.639: FB0 disp 0 max-fb 2 1920x1080 stride 3840 base 0x3f800000
007792.462: dtb_file 'bcm2712-rpi-5-b.dtb'
007889.387: dtparam: audio=on
007897.845: Unknown dtparam 'audio' - ignored
007955.877: Loaded overlay 'vc4-kms-v3d-pi5'
007958.842: dtparam: nvme=true
007972.827: dtparam: pciex1_gen=3
008353.118: RPM 6443, max RPM 6443
010485.395: Starting OS 10485 ms
010491.005: 00000040: -> 00000480
010492.854: 00000030: -> 00100080
010497.567: 00000034: -> 00100080
010502.280: 00000038: -> 00100080
010506.993: 0000003c: -> 00100080
010616.576: sdram: sdram refresh 2081->8324 (1)
038639.153: sdram: sdram refresh 2081->4162 (2)
070527.558: initial_turbo of 60 deactivated
```

I guess that the failure in the first iteration could happen between 10 and 38 second.

bukatlib, 2025-05-08:
I removed SDRAM_BANKLOW to use the default, note that my Pi 5 has 8 GB of memory, maybe the 16 GB version might slightly differ in memory timing.

popcornmix, 2025-05-08:
> I have noticed that a prime sieve might give me an incorrect count of primes mostly shortly after the boot.

Have you ever seen a failure on a second (or third, fourth etc) run of your code?
Is the failure only within the first N seconds of a boot?

bukatlib, 2025-05-08:
I have never observed a failure when the system heats up, it is either the first iteration and much less frequently the second. On the other hand, when the system is idle for a long time (cooler), it might be provoked as well (I remember to provoke it when I developed the code some time ago but thought that it is SW bug). I would bet that it is temperature related in some way.

pelwell, 2025-05-08:
Next time you run it, try with `initial_turbo=0` in config .txt.

popcornmix, 2025-05-08:
After next failure report output of `vcgencmd get_throttled`.

bukatlib, 2025-05-08:
Unfortunately I managed to reproduce the issue even with `initial_turbo=0` in config.txt but it was hard...
```
sieve size: 49999999999 bits (6250000000 bytes)
Sieve initialized in 1.1104 secs
Primes sieved in 14.9794 secs.
Found 4118054784 primes in 0.497368 secs (bandwidth 12566.1 MB/s).
Binary image written in 20.1927 secs.
Invalid fb3a27b0ea45ed9a82ea3c2c57b5d10791403f665c6661d8410224daadc26248 sum of sieve image.
```

```
# vcgencmd get_throttled
throttled=0x0
```

bukatlib, 2025-05-09:
Note that the program counted 29 less primes than it should but in the resulting image there is 52 additional bits cleared. It indicates a continuous deterioration in 64-byte cache line size.

bukatlib, 2025-05-11:
Note that the issue is probably the memory related, if it was the cache, I would expect that there would be multiple undercounted cache lines but there is just one. Would it be possible to have a testing EEPROM with a less strict memory timing and shorter memory refresh interval? Ideally, the temperature based optimization could be completely turned off. Thank you.

timg236, 2025-05-14:
> Note that the issue is probably the memory related, if it was the cache, I would expect that there would be multiple undercounted cache lines but there is just one. Would it be possible to have a testing EEPROM with a less strict memory timing and shorter memory refresh interval? Ideally, the temperature based optimization could be completely turned off. Note that my RPI 5 boots extremely fast (fast SSD), maybe it may be easier to reproduce it if it starts really fast. Thank you.

I think it should be possible to hack something in the firmware. No luck reproducing it at our end yet though

bukatlib, 2025-05-14:
If you have 16 GB RPi version, you might try to compute primes up to 240G:
```
#!/bin/bash

for i in $(seq 3); do
    ./prime_sieve_bench 240000000000 -
    sha256=$(openssl dgst -sha256 memory_primes.img | awk '{print $2}')
    if [ "${sha256}" != "72669d51616772bc710df6fe638d77610b8fa3a7124c84f62916c24f0407c615" ]; then
        mv memory_primes.img memory_primes_invalid_${i}.img
        echo "Invalid ${sha256} sum of sieve image." 1>&2
    else
        rm memory_primes.img
    fi
done

exit 0
```

I cannot run it on 8 GB RPi but on my x86_64 desktop I got this prime count:
```
sieve size: 119999999999 bits (15000000000 bytes)
Sieve initialized in 0.521748 secs
Primes sieved in 3.02842 secs.
Found 9538769484 primes in 0.225107 secs (bandwidth 66635.1 MB/s).
Binary image written in 22.1172 secs.
...
```

Adjustment of memory parameters could be nice to be configurable similar to some CPU parameters. It could allow to set only less strict timing or shorter refresh interval (similar to BIOS config found on x86 motherboards).

lurch, 2025-05-16:
> Adjustment of memory parameters could be nice to be configurable similar to some CPU parameters.

FYI there **are** some "memory parameters" - see https://www.raspberrypi.com/documentation/computers/config_txt.html#overclocking-options (although it looks like none of them can be changed on Pi 4 or Pi 5).

> (similar to BIOS config found on x86 motherboards)

Note that unlike an x86 motherboard, the RAM on a Raspberry Pi can't be swapped to a different DIMM 😉

timg236, 2025-05-19:
@bukatlib I've attached a bootloader which disables the firmware override for sdram refresh adjustment. Please could you try this?

It should help us determine whether this is caused by the cold->hot transition at boot
[pieeprom-no-sdram-refresh-adjust.zip](https://github.com/user-attachments/files/20281617/pieeprom-no-sdram-refresh-adjust.zip)

bukatlib, 2025-05-20:
I have just installed the new EEPROM and started to test.

The first observation is that the performance impact is negligible:
https://browser.geekbench.com/v6/cpu/11654514
https://browser.geekbench.com/v6/cpu/12053004

The sieve times seems to be more stable with the new EEPROM, I will keep testing in the following days...

timg236, 2025-05-20:
Thanks. That gives us food for thought. If we can’t reproduce it would you be open to sending the board back to us. We would send you replacement plus kit to compensate for postage / hassle

bukatlib, 2025-05-20:
How could I verify that it is correctly installed? This seems suspicious:
```
$ vcgencmd version
2025/05/08 15:13:17 
Copyright (c) 2012 Broadcom
version 69471177 (release) (embedded)
```

The following command was used to install it `rpi-eeprom-update -f pieeprom-no-sdram-refresh-adjust.bin`.

Thanks.

stu-spp, 2025-05-20:
> How could I verify that it is correctly installed? This seems suspicious:
>
> ```
> $ vcgencmd version
> 2025/05/08 15:13:17
> Copyright (c) 2012 Broadcom
> version 69471177 (release) (embedded)
> ```
>
> The [...]

```shell
pi@pi$ sudo vcgencmd bootloader_version
2025/05/08 15:13:17
version 69471177ba7e4cb7597cb2496f2a0b23f19c1113 (release)
timestamp 1746713597
update-time 1747729264
capabilities 0x0000007f

pi@pi$
```

timg236, 2025-05-21:
Sorry, the file is correct but because it was an incremental dev build from my local worktree the version info was not re-generated

Running `vclog -m ` with this version should NOT have any strings like this
`010679.132: sdram: sdram refresh 2081->4162 (2)`

At bootup / cold temperatures the firmware would push the SDRAM refresh settings into a higher performance band. It looks like on this board either that setting is unsafe or the system doesn't react quickly enough. I think we'll do some experiments to override the temperature bands on known good boards to see if we can figure out what the margin is.

Output of rpi-eeprom-update - note the May 19th date
```
pi@pi5:~ $ sudo rpi-eeprom-update -d -f pieeprom-no-sdram-refresh-adjust.bin
*** CREATED UPDATE pieeprom-no-sdram-refresh-adjust.bin  ***

   CURRENT: Thu  8 May 14:13:17 UTC 2025 (1746713597)
    UPDATE: Mon 19 May 09:58:27 UTC 2025 (1747648707)
    BOOTFS: /boot/firmware
'/tmp/tmp.POwwLmoKcN' -> '/boot/firmware/pieeprom.upd'

UPDATING bootloader. This could take up to a minute. Please wait

*** Do not disconnect the power until the update is complete ***

If a problem occurs then the Raspberry Pi Imager may be used to create
a bootloader rescue SD card image which restores the default bootloader image.

flashrom -p linux_spi:dev=/dev/spidev10.0,spispeed=16000 -w /boot/firmware/pieeprom.upd
Verifying update
VERIFY: SUCCESS
UPDATE SUCCESSFUL
```

bukatlib, 2025-05-21:
There is the output after the firmware update:
```
# vcgencmd bootloader_version
2025/05/08 15:13:17
version 69471177ba7e4cb7597cb2496f2a0b23f19c1113 (release)
timestamp 1746713597
update-time 1747763239
capabilities 0x0000007f
```

```
# vclog -m | grep sdram
009924.568: sdram: sdram refresh 2081->8324 (1)
462831.906: sdram: sdram refresh 2081->4162 (2)
463839.892: sdram: sdram refresh 2081->8324 (1)
465855.891: sdram: sdram refresh 2081->4162 (2)
```

```
# sha256sum pieeprom-no-sdram-refresh-adjust.bin 
2661acbe23509c688dbe0b770b99f1820044cd9db9d28027e3fc838775c12f11  pieeprom-no-sdram-refresh-adjust.bin
```

It is an unexpected output. Will try again...
```
# rpi-eeprom-update -d -f pieeprom-no-sdram-refresh-adjust.bin
*** CREATED UPDATE pieeprom-no-sdram-refresh-adjust.bin  ***

   CURRENT: Čt 8. května 2025, 14:13:17 UTC (1746713597)
    UPDATE: Po 19. května 2025, 09:58:27 UTC (1747648707)
    BOOTFS: /boot/firmware
'/tmp/tmp.l9VWmcRRGp' -> '/boot/firmware/pieeprom.upd'

UPDATING bootloader. This could take up to a minute. Please wait

*** Do not disconnect the power until the update is complete ***

If a problem occurs then the Raspberry Pi Imager may be used to create
a bootloader rescue SD card image which restores the default bootloader image.

flashrom -p linux_spi:dev=/dev/spidev10.0,spispeed=16000 -w /boot/firmware/pieeprom.upd
Verifying update
VERIFY: SUCCESS
UPDATE SUCCESSFUL
```

bukatlib, 2025-05-21:
Avoided to edit EEPROM config and retested the performance impact on GeekBench6:
https://browser.geekbench.com/v6/cpu/11654514
https://browser.geekbench.com/v6/cpu/12068611

Performance overhead is 0.5 % singlethreaded and 1.7 % multithreaded.

My view is that the temperature based tuning is pretty tricky, it depends on many factors such as the external temperature, fan characteristics, case, fan, position of sensors, workloads, ... Overoptimistic constants might have fatal consequences; just image that some kernel memory range is randomly zeroed. I suggest to add SDRAM_REFRESH_TEMP_ADJUST config option or even better to completely remove temperature based adjustments. I am not sure whether up to 1.7% performance speedup is worthy the risks (kernel hangs, freezes, app crashes, ...) for a miniaturized embedded computer that might be used in the industry.

I will run many tests with the new firmware tomorrow when I have more time. Many thanks for all the feedback.

bukatlib, 2025-05-21:
How can I avoid automatic updates of EEPROM? I have just started RPI from power-off state and the firmware was flashed to the original one... It makes any testing hard. Thank you.

lurch, 2025-05-21:
> How can I avoid automatic updates of EEPROM?

See https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#automaticupdates

bukatlib, 2025-05-22:
Thank you, masking systemctl service helped. Note that the firmware without the memory refresh adjust can be easily detected from the prime sieve run since counting bits in GBs of data is about 5% slower (throughput ~12100 vs 12800 MB/s)... A few runs without errors, will continue testing...

timg236, 2025-05-22:
Thanks for the update. 

Re updates: If you want to use custom EEPROM config settings with an 'out of tree' image you can do the following
`sudo rpi-eeprom-config --edit pieeprom-no-sdram-refresh-adjust.bin`

The 'low temperature' mode is within the JEDEC standard limits so our investigation is focussed on whether some boards could have less temperature margin here than others, or there is a software bug. It will take a little while but I think your test program could be a useful test for us in temperature controlled setups to help us fully understand what's happening.

bukatlib, 2025-05-29:
@timg236 I can send the device assembled in Argon NEO M.2 NVME case without SSD hard drive (Lexar NM710 1 TB). It would maximise your chances to reproduce it. Maybe just an idea, the case included cooling pads that are smaller than the memory chip area, it would be possible that the temperature distribution across the memory chip area would not be uniform causing the issue when the sensor measures lower temperature at the center (covered by a cooling pad) than it is reality at chip's distant edges.  I would prefer 16 GB version instead of extras if possible, the best would be a piece with 64-bit memory bus (instead of 32-bit bus width) but it is surely infeasible for RPI5 :-).

timg236, 2025-05-29:
> [@timg236](https://github.com/timg236) I can send the device assembled in Argon NEO M.2 NVME case without SSD hard drive (Lexar NM710 1 TB). It would maximise your chances to reproduce it. Maybe [...]

It might be better to send the bare board, we'll need to take it out the case anyway and put it in a temperature controlled environment where we can apply heat/cooling to DDR directly.
Happy to send replacement plus an additional Pi5 16GB but can't add more wires to bus :) 
Please can you send an e-mail to info@raspberrypi.com with details - so long as we can make customers / taxes work it should be fine.

bukatlib, 2025-06-17:
@timg236 Have you managed to reproduce it on the returned board with the default firmware? Thanks.

timg236, 2025-06-18:
So far, no. Testing with a large USB fan it didn't reproduce the issue.

It's currently soak testing with an active cooler, run for 5 iterations - cooldown reboot - repeat.

bukatlib, 2025-07-02:
Testing every day a few times on a new RPI 16GB with the same steel case, NVME, and SW setup and so far not reproduced. Given my previous experience and frequency of faults, I would already expect to see it a few times if it were defective. This shows two things; the first is that even in the lab environment it might be hard to reproduce since the HW+SW configuration differ (swapping, cooling/heating, boot time, ...), the second is that the memory timing "might not be too tight" but RPI 8GB is probably faulty. I wonder whether the memory should be cooled or heated or both in the lab environment since the steel case is a "thermal bridge" between CPU and main memory. Unfortunately I did not send the board in the original case since reproducing it in close to the original configuration should be easier.

lurch, 2025-07-03:
@bukatlib Just out of curiosity: were you ever able to reproduce this problem with the Pi outside of your steel case, or is that a configuration that you never tested?

timg236, 2025-07-03:
> Testing every day a few times on a new RPI 16GB with the same steel case, NVME, and SW setup and so far not reproduced. Given my previous experience and frequency of faults, I would already expect to see it a few times if it were defective. This shows two things; the first is that even in the lab environment it might be hard to reproduce since the HW+SW configuration differ (swapping, cooling/heating, boot time, ...), the second is that the memory timing "might not be too tight" but RPI 8GB is probably faulty. I wonder whether the memory should be cooled or heated or both in the lab environment since the steel case is a "thermal bridge" between CPU and main memory. Unfortunately I did not send the board in the original case; reproducing it in close to the original configuration should be easier.

Did you manage to reproduce this on the replacement 8GB? I don't think there is an issue with the memory timing as such but possibly this particular board has less margin, it's not failed for us yet using an official Pi5 power supply and an active cooler

bukatlib, 2025-09-08:
If the memory is defective, the physical addresses would reveal whether leaked memory cells are close to each other (based on the arrangement of memory rows and columns) or distribution pattern if any. It could also indicate a place where the discharge occurs on the chip. Porting to u-boot would be pretty hard, I can imagine to rewrite it to C and use some compiler helpers for bit operations but I doubt that multi-threading would be easy to accomplish in the bootloader, it is a pretty limited environment. The reason why I suspect memory rather than the CPU is that even with the virtual addresses, the zeroed bits are inside a small address range (in the sieve image) and the leakage was probably a longer process since the program undercounted primes but wrote even less primes later (observed only a few times).

timg236, 2025-09-08:
Did the error ever occur again on the new boards?

timg236, 2026-10-02:
Closed as can't reproduce -  I'm still using the RMA Pi as my daily driver on Trixie

closed 2026-10-02 by timg236