raspberrypi/rpi-eeprom #705: 2711 eeprom firmware having issues when using tryboot with partition table not in disk order

berkutta, 2025-05-08:
### Describe the bug

I have the following partition table:
```
Disk /dev/sdc: 58.94 GiB, 63281561600 bytes, 123596800 sectors
Disk model: USB3.0 CRW   -SD
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x3703e39f

Device     Boot   Start       End   Sectors  Size Id Type
/dev/sdc1         16384   1064959   1048576  512M  c W95 FAT32 (LBA)
/dev/sdc2       3162112 123596799 120434688 57.4G 83 Linux
/dev/sdc3       1064960   2113535   1048576  512M  b W95 FAT32
/dev/sdc4       2113536   3162111   1048576  512M  b W95 FAT32

Partition table entries are not in disk order.
```
sdc1=trybootfs, sdc3=bootfsa, sdc4=bootfsb

Setting boot_partition to 3 in autoboot.txt works, setting it to 4 doesn't work.

Log output with boot_partition=4:
```
  4.69 [sdcard] vl805.bin not found
  4.70 [sdcard] pieeprom.upd not found
  4.70 [sdcard] recover4.elf not found
  4.70 [sdcard] recovery.elf not found
  4.30 Read start4.elf bytes  2263968 hnd 0x5487
  4.34 Read fixup4.dat bytes     5456 hnd 0x257
  4.36 0x00c03111 0x00000000 0x00001fff
  4.40 MEM GPU: 76 ARM: 948 TOTAL: 1024
  4.73 Starting start4.elf @ 0xfec00200 partition 4
  4.87 XHCI-STOP
  4.87 xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
  4.88 USBSTS 18
  4.88 PCI0 reset
  4.89 +

MESS:00:00:04.996169:0: arasan: arasan_emmc_open
MESS:00:00:04.997830:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:05.118063:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:05.130933:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:05.173394:0: boot-part: 4 fs-type: 0
MESS:00:00:05.174805:0: boot-part: 4 fs-type: 0
MESS:00:00:05.179185:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:05.299223:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:05.312107:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:05.354433:0: boot-part: 4 fs-type: 0
MESS:00:00:05.355848:0: boot-part: 4 fs-type: 0
MESS:00:00:05.360228:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:05.376266:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:05.486933:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:05.499367:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:05.512232:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:05.551401:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_WRITE_PROTECTED before we're ready
MESS:00:00:05.563709:0: boot-part: 4 fs-type: 0
MESS:00:00:05.565119:0: boot-part: 4 fs-type: 0
MESS:00:00:05.569499:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:05.585300:0: gpioman: gpioman_get_pin_num: attempting to get pin number for SDCARD_CONTROL_POWER before we're ready
MESS:00:00:05.618124:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.622634:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:05.639781:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.644284:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.649518:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:05.757350:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:05.766661:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:05.779544:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
[... 262 lines cut ...]
/dev/mmcblk0p4      3162112 123596799 120434688 57.4G 83 Linux
```

### Steps to reproduce the behaviour

- Flash Raspbian 64bit lite to original 64GB Raspberry Pi SD-Card
- Use gparted to shrink rootfs, copy bootfs two times in between
- Create autoboot.txt on first bootfs

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

```
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
FREEZE_VERSION=0
```

### System

[raspinfo.txt](https://github.com/user-attachments/files/20111576/raspinfo.txt)

timg236, 2025-05-09:
I spotted something and have attached a highly experimental fix candidate (replace start.elf / fixup.dat files with the attached)

[part4.zip](https://github.com/user-attachments/files/20128495/part4.zip)

berkutta, 2025-05-10:
> I spotted something and have attached a highly experimental fix candidate (replace start.elf / fixup.dat files with the attached)

Nice, all four test cases working! Thanks for the quick analyze and fix.
Always crazy to see what Raspberry Pi's are capable to do with a bit of new Firmware (and even decade old ones getting the same features)!

---

## Test with ordered table (old start4/fixup file)
```
Device     Boot   Start       End   Sectors  Size Id Type
/dev/sdc1         16384   1064959   1048576  512M  c W95 FAT32 (LBA)
/dev/sdc2       1064960   2113535   1048576  512M  c W95 FAT32 (LBA)
/dev/sdc3       2113536   3162111   1048576  512M  c W95 FAT32 (LBA)
/dev/sdc4       3162112 123596799 120434688 57.4G 83 Linux
```
- boot with boot_partition=3 working
- boot with boot_partition=2 working

```
$ vcgencmd version
Apr 30 2025 13:33:39 
Copyright (c) 2012 Broadcom
version 5560078dcc8591a00f57b9068d13e5544aeef3aa (clean) (release) (start)
```

---

## Test with NOT ordered table (old start4/fixup file)
```
Device     Boot   Start       End   Sectors  Size Id Type
/dev/sdc1         16384   1064959   1048576  512M  c W95 FAT32 (LBA)
/dev/sdc2       3162112 123596799 120434688 57.4G 83 Linux
/dev/sdc3       1064960   2113535   1048576  512M  c W95 FAT32 (LBA)
/dev/sdc4       2113536   3162111   1048576  512M  c W95 FAT32 (LBA)
```

- boot with boot_partition=3 working
- boot with boot_partition=4 NOT working with repeating error output like in issue description
- _boot with boot_partition=2 also not working, not UART output, I guess due to start4 file not loadable as this is ext4 rootfs_

```
$ vcgencmd version
Apr 30 2025 13:33:39 
Copyright (c) 2012 Broadcom
version 5560078dcc8591a00f57b9068d13e5544aeef3aa (clean) (release) (start)
```

---

## Test with ordered table (new start4/fixup file)
```
Device     Boot   Start       End   Sectors  Size Id Type
/dev/sdd1         16384   1064959   1048576  512M  c W95 FAT32 (LBA)
/dev/sdd2       1064960   2113535   1048576  512M  c W95 FAT32 (LBA)
/dev/sdd3       2113536   3162111   1048576  512M  c W95 FAT32 (LBA)
/dev/sdd4       3162112 123596799 120434688 57.4G 83 Linux
```

- boot with boot_partition=2 working
- boot with boot_partition=3 working

```
$ vcgencmd version
May  9 2025 20:41:07 
Copyright (c) 2012 Broadcom
version 0201939dda3c4f8344d16670766bd80749bd8a8b (clean) (release) (start)
```

---

## Test with NOT ordered table (new start4/fixup file)
```
Device     Boot   Start       End   Sectors  Size Id Type
/dev/sdd1         16384   1064959   1048576  512M  c W95 FAT32 (LBA)
/dev/sdd2       3162112 123596799 120434688 57.4G 83 Linux
/dev/sdd3       1064960   2113535   1048576  512M  c W95 FAT32 (LBA)
/dev/sdd4       2113536   3162111   1048576  512M  c W95 FAT32 (LBA)
```

- boot with boot_partition=3 WORKING
- boot with boot_partition=4 working

```
$ vcgencmd version
May  9 2025 20:41:07 
Copyright (c) 2012 Broadcom
version 0201939dda3c4f8344d16670766bd80749bd8a8b (clean) (release) (start)
```

pelwell, 2025-05-10:
You can close the issue if you are happy with the outcome.

closed 2025-05-10 by berkutta