raspberrypi/rpi-eeprom #706: By default, USB should always take precedent over NVMe in boot order

ell1e, 2025-05-10:
### Describe the bug

Perhaps it's just me, but I feel like by default, USB should always take precedent over NVMe in boot order even once an OS was installed onto the NVMe.

There's a simple reason why I'm thinking that: it's how all desktops and laptops nowadays work, because if you explode your on-disk install such that the OS bootloader still recognizes and starts it, but it doesn't reach a desktop anymore, you need to be able to boot some live usb stick to fix it.

This seems even more important on the Raspberry Pi 5, since unlike on a desktop UEFI pc, I can hammer Escape all I want and there's apparently no boot screen and no boot setup and no boot device choice popup that seems to be reachable that way. This seems to mean that in practice, the device can effectively end up bricked unless the NVMe is fully removed (which is difficult since it's usually screwed in), or this default boot order is changed e.g. while the NVMe installed system is still operational, which obviously can't be done anymore once it's bricked like that. A USB over NVMe default would completely avoid this problem.

Sadly, not only is USB first not the default, but it seems like the boot order to fulfill that suggestion (which would either be USB, SD card, NVMe, or alternatively USB, NVMe, SD acrd) doesn't even seem to be a possible option even if the user wanted to:

[image]

### Steps to reproduce the behaviour

1. Install a PCIe M2 HAT on Raspberry Pi 5, e.g. the official one. Make sure an NVMe SSD M2 drive is inserted.
2. Install Raspberry Pi OS on the NVMe drive.
3. Shutdown Raspberry Pi 5.
4. Insert a bootable USB stick with a Raspberry OS bookworm default live image
5. Start the Raspberry Pi 5.
6. Hammer escape key and try all means possible to boot from the USB without relying on the NVMe system to work. This doesn't seem to be possible. No matter what you do, you'll end up with the NVMe system, even if that one happens to be bricked.
7. Now check out the documentation to change the boot order https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#boot_order-examples and it doesn't seem like USB before NVMe is even available, let alone the default.

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

It seems like all possible configurations are affected, at least if the [documentation is correct](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#boot_order-examples). It concerns both the suboptimal default boot order, as well as the apparently lacking option to even pick one that fulfills this behavior that most UEFI desktop computers have.

### System

Affects all operating systems

lurch, 2025-05-10:
> I can hammer Escape all I want and there's apparently no boot screen and no boot setup and no boot device choice popup that seems to be reachable that way.

If your Pi 5 firmware is reasonably up to date, you can press the Space bar just after power-on to get a boot-device selection menu. See https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2712/release-notes.md#2024-10-10-add-support-to-override-the-boot-mode-at-power-on-latest

> It seems like all possible configurations are affected, at least if the [documentation is correct](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#boot_order-examples).

As the title you've linked to suggests, those are merely _examples_; more fully-fledged documentation is in the preceding section https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#boot_order-fields
So to fulfil your request of "USB, SD card, NVMe" you'd just need to set your `BOOT_ORDER` to `0xf614`.

ell1e, 2025-05-10:
Thank you so much for your answer! I would suggest 0xf614 should be the default then. Many users will not think of that until it is too late, and then removing the NVMe can be quite involved.

lurch, 2025-05-11:
I'm not directly involved with the rpi-eeprom code myself, but I suspect that the reason that USB boot _isn't_ the default, is that enumerating the USB devices connected to a Pi takes additional time, which is wasted boot-time if you don't have any bootable USB MSDs attached.
Note that as well as [manually editing](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-bootloader-configuration) the `BOOT_ORDER`, you can also change it using [`raspi-config`](https://www.raspberrypi.com/documentation/computers/configuration.html#advanced-options), or by selecting one of the "Misc utility images" -> "Bootloader" options in [Raspberry Pi Imager](https://www.raspberrypi.com/software/).

timg236, 2025-05-11:
@lurch is correct about the reasoning for the defaults. Closing.

closed 2025-05-11 by timg236

ell1e, 2025-05-12:
I don't mean to intrude, I'd just like to leave for anybody interested the argument that even with above reasoning, it can be argued that USB first might be the better default.

I tested out the boot delay, it's substantial but in my opinion still worth to not be bricked without literally dismantling the system, and it matches the default behavior of most other desktop systems out there. I assume that is what most Raspberry Pi 5 users might expect out of the box. And it's still possible to change the order for faster boots.

But I get that y'all have other things to focus on than this topic.

ell1e, 2025-05-12:
> nd up with the system bricked without literally dismantling it, 

That is what I meant to say with that line. If you use PCIe NVMe M2 and a case, it can be quite difficult and require dismantling half the thing to swap it out. E.g. the PCIe hat I use has to be completely taken apart to swap the SSD, since the screw is facing toward the inside. Something similar would be the case e.g. for a desktop PC with a case.

ell1e, 2025-05-12:
On a related but separate note, I wonder, is there a reason why escape or del don't lead to the boot menu prompt? I think that's a more common key and might make it easier to discover.

pelwell, 2025-05-12:
> On a related but separate note

That's not how this forum works.

>  is there a reason why escape or del don't lead to the boot menu prompt? I think that's a more common key and might make it easier to discover.

No discovery required, just reading - the screen tells you what to press. I find both keys easier to press than ESC or DEL.

ell1e, 2025-05-12:
Sorry, I'll stop responding since this seems to be the wrong place. But perhaps it's fair if I still point out that I don't see any prompt since the firmware boot screen seems to be shorter than my physical screen reacts to the new signal.

lurch, 2025-05-12:
> E.g. the PCIe hat I use has to be completely taken apart to swap the SSD, since the screw is facing toward the inside.

In that case you might be happy to see the "knurled double-flanged drive attachment screw" we use on the official [M.2 HAT+](https://www.raspberrypi.com/documentation/accessories/m2-hat-plus.html#m2-hat-plus) 😂 (which doesn't even need any tools)

timg236, 2025-05-20:
You can override this via the NET_INSTALL_KEYBOARD_WAIT eeprom config setting 

The defaults are selected for the common use case ie simple usb

ell1e, 2025-05-20:
My apologies if there's a misunderstanding, I agree that NVMe probably isn't a common use case, which is why I would suggest not ever defaulting to it over USB.

pelwell, 2025-05-20:
> I agree that NVMe probably isn't a common use case

What gives you that idea?

timg236, 2025-05-20:
The BOOT_ORDER discussion is settled.
We aren’t changing the defaults for everyone.

If there is an issue with particular keyboards not working then I’m happy to have a new bug for that.  The rule of thumb is that if we can make it work out the box without making things worse for the common case then we do that. Otherwise it’s down to EEPROM config quirks eg boot delay for spinning rust

lurch, 2025-05-21:
> But in my case neither the USB keyboard nor the screen seem to work in time for any sort of boot prompt to be possibly read or interacted with

Just to clarify Tim's earlier message: you can use the https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#NET_INSTALL_KEYBOARD_WAIT EEPROM setting to make the bootloader wait longer for a keyboard to be detected, before continuing with the rest of the boot.