raspberrypi/rpi-eeprom #709: RPi 5 with official case fan always at 100%

floviolleau, 2025-05-19:
### Describe the bug

I have a Raspberry Pi 5 Model B Rev 1.0, 8GB and the official Rpi case box with the fan. A few seconds after applying power, the fan goes to 100% full speed and never stops despite it wasn't the case since the last reboot.

[image]

The last reboot was a long time ago since no reboot something around february (it is a server so I cannot remember the last reboot). 

My issue seems to be similar to https://github.com/raspberrypi/rpi-eeprom/issues/660, https://github.com/raspberrypi/rpi-eeprom/issues/678 and https://github.com/raspberrypi/rpi-eeprom/issues/697.

In my case :
- `dtparam=cooling_fan=on` made no difference
- I also tried `sudo dtparam cooling_fan` as suggested in the other issue but it had no effect
- Cold boot or soft reboot makes no difference in my case
- `pinctrl FAN_PWM op dh` stop the fan
- `pinctrl FAN_PWM op dl` start the fan

I've tried to include all of the debug information that has been asked for in the other issues - please let me know what else I can provide.

## vcgencmd measure_temp

`temp=37.3'C`

## pinctrl | grep FAN

```
29: no    pu | hi // FAN_TACH/GPIO29 = none
45: a0    pd | lo // FAN_PWM/GPIO45 = PWM1_CHAN3
```

## grep . /sys/class/hwmon/hwmon*/{temp,fan,pwm}*

```
/sys/class/hwmon/hwmon0/temp1_input:43550
grep: /sys/class/hwmon/hwmon*/fan*: No such file or directory
grep: /sys/class/hwmon/hwmon*/pwm*: No such file or directory
```

## grep -s . /sys/class/thermal/thermal_zone*/* /sys/class/thermal/cooling_device*/*

```
/sys/class/thermal/thermal_zone0/available_policies:step_wise 
/sys/class/thermal/thermal_zone0/integral_cutoff:0
/sys/class/thermal/thermal_zone0/k_d:0
/sys/class/thermal/thermal_zone0/k_i:0
/sys/class/thermal/thermal_zone0/k_po:0
/sys/class/thermal/thermal_zone0/k_pu:0
/sys/class/thermal/thermal_zone0/mode:enabled
/sys/class/thermal/thermal_zone0/offset:450000
/sys/class/thermal/thermal_zone0/policy:step_wise
/sys/class/thermal/thermal_zone0/slope:-550
/sys/class/thermal/thermal_zone0/sustainable_power:0
/sys/class/thermal/thermal_zone0/temp:36950
/sys/class/thermal/thermal_zone0/trip_point_0_hyst:0
/sys/class/thermal/thermal_zone0/trip_point_0_temp:110000
/sys/class/thermal/thermal_zone0/trip_point_0_type:critical
/sys/class/thermal/thermal_zone0/trip_point_1_hyst:5000
/sys/class/thermal/thermal_zone0/trip_point_1_temp:50000
/sys/class/thermal/thermal_zone0/trip_point_1_type:active
/sys/class/thermal/thermal_zone0/trip_point_2_hyst:5000
/sys/class/thermal/thermal_zone0/trip_point_2_temp:60000
/sys/class/thermal/thermal_zone0/trip_point_2_type:active
/sys/class/thermal/thermal_zone0/trip_point_3_hyst:5000
/sys/class/thermal/thermal_zone0/trip_point_3_temp:67500
/sys/class/thermal/thermal_zone0/trip_point_3_type:active
/sys/class/thermal/thermal_zone0/trip_point_4_hyst:5000
/sys/class/thermal/thermal_zone0/trip_point_4_temp:75000
/sys/class/thermal/thermal_zone0/trip_point_4_type:active
/sys/class/thermal/thermal_zone0/type:cpu-thermal
```
[... 608 lines cut ...]
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

</details

### NVMe boot

N/A

### Network (TFTP boot)

N/A

floviolleau, 2025-05-19:
Because my docker install throwed errors:

```
mai 19 02:03:06 main-server-2 dockerd[88551]: time="2025-05-19T02:03:06.932543044+02:00" level=info msg="Starting up"
mai 19 02:03:06 main-server-2 dockerd[88551]: time="2025-05-19T02:03:06.933005101+02:00" level=info msg="OTEL tracing is not configured, using no-op tracer provider"
mai 19 02:03:06 main-server-2 dockerd[88551]: time="2025-05-19T02:03:06.940235167+02:00" level=info msg="Creating a containerd client" address=/run/containerd/containerd.sock timeout=1m0s
mai 19 02:03:06 main-server-2 dockerd[88551]: time="2025-05-19T02:03:06.951220392+02:00" level=error msg="failed to mount overlay: no such device" storage-driver=overlay2
mai 19 02:03:06 main-server-2 dockerd[88551]: time="2025-05-19T02:03:06.951255042+02:00" level=error msg="[graphdriver] prior storage driver overlay2 failed: driver not supported"
mai 19 02:03:06 main-server-2 dockerd[88551]: failed to start daemon: error initializing graphdriver: driver not supported
mai 19 02:03:06 main-server-2 systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
```

I remembered that I switched to the firmware updater by Hexxeh, enhanced by AndrewS and Dom.

So, I did a `sudo rpi-update`:

```
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:d688b628e142f64cde5ceabd9259f813ccae77d2
BOOTLOADER_REV:b08dff7e46329af656ea93bb8ab504a68c39322d
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Remove old modules backup
 *** Backing up modules 6.12.25+rpt-rpi-2712
WANT_32BIT:0 WANT_64BIT:1 WANT_64BIT_RT:0 WANT_PI4:1 WANT_PI5:1
##############################################################
WARNING: This update bumps to rpi-6.12.y linux tree
See discussions at:
https://forums.raspberrypi.com/viewtopic.php?t=379745
##############################################################
Would you like to proceed? (y/N)
Downloading bootloader tools
Downloading bootloader images
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  152M  100  152M    0     0  37.2M      0  0:00:04  0:00:04 --:--:-- 41.5M
BOOTLOADER: up to date
   CURRENT: jeu. 08 mai 2025 14:13:17 UTC (1746713597)
    LATEST: jeu. 08 mai 2025 14:13:17 UTC (1746713597)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 6.12.28-v8-16k+
 *** depmod 6.12.28-v8+
 *** Updating VideoCore libraries
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to d688b628e142f64cde5ceabd9259f813ccae77d2
 *** A reboot is needed to activate the new firmware
```

Now my Docker is working fine:
```
● docker.service - Docker Application Container Engine
     Loaded: loaded (/lib/systemd/system/docker.service; enabled; preset: enabled)
     Active: active (running) since Mon 2025-05-19 02:05:26 CEST; 1min 18s ago
TriggeredBy: ● docker.socket
       Docs: https://docs.docker.com
   Main PID: 1347 (dockerd)
      Tasks: 83
        CPU: 866ms
     CGroup: /system.slice/docker.service
             ├─1347 /usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock
             ├─2356 /usr/bin/docker-proxy -proto tcp -host-ip <redacted> -host-port 8079 -container-ip 172.17.0.2 -container-port 80 -use-listen-fd
             ├─2381 /usr/bin/docker-proxy -proto tcp -host-ip <redacted> -host-port 3012 -container-ip 172.17.0.2 -container-port 3012 -use-listen-fd
             ├─2416 /usr/bin/docker-proxy -proto tcp -host-ip <redacted> -host-port 8078 -container-ip 172.17.0.3 -container-port 8080 -use-listen-fd
             ├─2468 /usr/bin/docker-proxy -proto tcp -host-ip <redacted> -host-port 8077 -container-ip 172.17.0.4 -container-port 80 -use-listen-fd
             ├─2554 /usr/bin/docker-proxy -proto tcp -host-ip <redacted> -host-port 8015 -container-ip 172.18.0.2 -container-port 3000 -use-listen-fd
             ├─2565 /usr/bin/docker-proxy -proto tcp -host-ip <redacted> -host-port 8015 -container-ip 172.18.0.2 -container-port 3000 -use-listen-fd
             ├─2651 /usr/bin/docker-proxy -proto tcp -host-ip <redacted> -host-port 8016 -container-ip 172.18.0.4 -container-port 80 -use-listen-fd
             └─2659 /usr/bin/docker-proxy -proto tcp -host-ip <redacted> -host-port 8016 -container-ip 172.18.0.4 -container-port 80 -use-listen-fd
```

AND now the fan is not at 100%........ O_o

So problem seems to be solved. I let it open for a week in case, if it wasn't the case.

Now `sudo cat /sys/kernel/debug/pwm` return a correct output
```
0: platform/1f0009c000.pwm, 4 PWM devices
 pwm-0   ((null)              ): period: 0 ns duty: 0 ns polarity: normal
 pwm-1   ((null)              ): period: 0 ns duty: 0 ns polarity: normal
 pwm-2   ((null)              ): period: 0 ns duty: 0 ns polarity: normal
 pwm-3   (cooling_fan         ): requested enabled period: 41566 ns duty: 12225 ns polarity: inverse usage_power
```

before:
```
0: platform/1f0009c000.pwm, 4 PWM devices
 pwm-0   ((null)              ): period: 0 ns duty: 0 ns polarity: normal
 pwm-1   ((null)              ): period: 0 ns duty: 0 ns polarity: normal
 pwm-2   ((null)              ): period: 0 ns duty: 0 ns polarity: normal
 pwm-3   ((null)              ): period: 0 ns duty: 0 ns polarity: normal
```

timg236, 2025-05-19:
The bootloader is responsible for auto-detecting the fan but it's operation after the OS has started. It does this by driving PWM and then monitoring fan-tach. If a fan is detected the dtoverlay is loaded.
It sounds like this was not happening but if it's working after rpi-udpate then it's likely that the change to probe both max-rpm and current-rpm resolved this for you you
https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2712/release-notes.md#2025-04-07-arm_dt-revert-to-using-the-max-fan-speed-latest.

To check the fan probe run `sudo vclog -m` and look for lines like `RPM 5009, max RPM 5009"

timg236, 2025-06-23:
If setting "dtparam=cooling_fan=on" does not resolve the issue, then unfortunately, I think this is a hardware issue e.g. PWM connector pin is shorted to ground, and this can't be resolved with a software change.

closed 2025-06-23 by timg236