raspberrypi/rpi-eeprom #713: Higher CPU temperature after update to 2025-05-08

danez, 2025-05-27:
### Describe the bug

I noticed that after updating from 2025-03-10 to 2025-05-08 the CPU temperature is behaving differently. It is slightly higher, and therefore also the fan ramps up a lot more. Before the update, you can see a lot of dips when at night the window was open and the RaspberryPI 5 cools down, as it is located on the floor.

I then reverted to 2025-03-10 yesterday and it already looks like it is back to what it was before. And also the fan does not spin up as much.

[image]

* Fan speed is read by doing `cat /sys/devices/platform/cooling_fan/hwmon/*/fan1_input` every 10 seconds.

Now, of course, I was also wondering if maybe just the shutdown of HAOS and rebooting might have already fixed something by itself.

Did anybody else notice something similar?

I have my production HAOS running again now and don't want to unnecessarily boot to RaspiOS, but if there is anything I can provide, let me know.

### Steps to reproduce the behaviour

Update to 2025-05-08

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

Default for Raspi 5, except bootorder changes for NVMe boot

### System

OS: HAOS 15.2
bootloader: 2025-05-08

timg236, 2025-05-28:
There doesn't appear to be any interesting changes in clocks or power management code between those two releases. To confirm you could try running the bcmstat tool to observe changes to clock frequencies and core voltage whilst the system is idle
https://github.com/popcornmix/bcmstat

danez, 2025-06-12:
So over the last 2 weeks it fluctuated a little bit more, but the temperature stayed high, cpu freq are identical. So I assume the "issue" is probably due to code changes in home assistant or somewhere else and NOT the bootloader. Thanks for the replies though.

closed 2025-06-12 by danez

timg236, 2025-06-12:
Thanks for replying with an update.

For reference, another useful diagnostic is 'vcgencmd pmic_read_adc' which will show the voltage and current on the PMIC outputs (but sadly not USB VBUS).