raspberrypi/rpi-eeprom #719: Pasbberry Pi 5 OVERLAP DETECTED! after "program_pubkey=1"

Xamarin2019, 2025-06-25:
### Describe the bug

After programming Secure-boot with the configuration `proogram_pubkey = 1`  Raspberry Pi 5 gives an error:
```
[    0.000000] OF: reserved mem: OVERLAP DETECTED!
[    0.000000] atf@0 (0x0000000000000000--0x0000000000080000) overlaps with nvram@0 (0x0000000000038000--0x00000000000382c3)
```

It is also incorrectly read rpi-eeprom-config:
`Traceback (most recent call last):
  File "/usr/bin/rpi-eeprom-config", line 607, in <module>
    main()
  File "/usr/bin/rpi-eeprom-config", line 599, in main
    current_config, config_src = read_current_config()
                                 ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/rpi-eeprom-config", line 234, in read_current_config
    return (open(nvmem_base + d + "/nvmem", "rb").read().decode('utf-8'), "blconfig device")
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x9c in position 1: invalid start byte
`

### Steps to reproduce the behaviour

Programm Secure-boot with the configuration `proogram_pubkey = 1`  on Raspberry Pi 5

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

 rpi-eeprom-config
```
Traceback (most recent call last):
  File "/usr/bin/rpi-eeprom-config", line 607, in <module>
    main()
  File "/usr/bin/rpi-eeprom-config", line 599, in main
    current_config, config_src = read_current_config()
                                 ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/rpi-eeprom-config", line 234, in read_current_config
    return (open(nvmem_base + d + "/nvmem", "rb").read().decode('utf-8'), "blconfig device")
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x9c in position 1: invalid start byte
```

### System

[raspinfo.txt](https://github.com/user-attachments/files/20904216/raspinfo.txt)

cat /etc/rpi-issue
```
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, b15527e15ba4d9e65126a454bfec2040b909dc1c, stage2
```

vcgencmd bootloader_version
```
2025/02/12 10:51:52
version f788aab60121515fe59c37c899186831e6c19585 (release)
timestamp 1739357512
update-time 1747297204
capabilities 0x000000ff
```
vcgencmd version
```
2025/02/12 10:51:52
Copyright (c) 2012 Broadcom
version f788aab6 (release) (embedded)
```

uname -a
```
Linux raspberrypi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
```

timg236, 2025-06-25:
Please can you attach the EEPROM binary that you supplied to rpiboot

Xamarin2019, 2025-06-26:
I used USBBOOT and there is a link in it https://github.com/raspberrypi/usbboot/blob/master/secure-boot-recovery5/recovery.original.binwhich leads to rpi-eeprom/firmware-2712/latest/recovery.bin

I also wanted to find a solution on my own and tried different versions of recovery.bin, but the error still present.

timg236, 2025-06-26:
Actually, I've managed to reproduce this

Xamarin2019, 2025-06-26:
> Could you attach the binary being flashed instead of the source image?
> 
> ```
> cd secure-boot-recovery5
> # export KEY_FILE=your-private-key.pem
> ../tools/update-pieeprom.sh -f -k "${KEY_FILE}"
> ```
> 
> Attach `pieeprom.bin`

[pieeprom.bin.txt](https://github.com/user-attachments/files/20927618/pieeprom.bin.txt)

Xamarin2019, 2025-06-27:
```
raspberry5@raspberrypi:~ $ vcgencmd bootloader_version
2025/06/26 16:09:11
version cf73e23d0e9b28236449a4fc9fe26ad6ce2344aa (release)
timestamp 1750950551
update-time 1751018492
capabilities 0x000000ff

raspberry5@raspberrypi:~ $ rpi-eeprom-config
[all]
BOOT_UART=0
SIGNED_BOOT=1
POWER_OFF_ON_HALT=1

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
# Try SD first (1), followed by, USB (RP1), NVMe PCIe, then network
BOOT_ORDER=0xf41

# Disable self-update mode - to prevent automatic updates of unsigned bootloader images
ENABLE_SELF_UPDATE=0

# Turn off HDMI during boot
DISABLE_HDMI=1

```

It looks like the problem is fixed, I put it on my testing.

Thanks a lot! 
It is very pleasant when they answer so quickly!

closed 2025-08-01 by Xamarin2019