raspberrypi/rpi-eeprom #728: RFE rPi5: Need Boot Media Selection Menu On Build in Serial (ttyAMA10)

marcmerlin, 2025-07-07:
I have an rPI5 with bootable sdcard and nvme (both are confirmed working).
Currently I have this
```
aragorn:~# rpi-eeprom-config
[all]
CONSOLE=UART
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf1654
#BOOT_ORDER=0xf16
PCIE_PROBE=0
```

Adding 45 first in the boot causes the boot to be delayed and I do get the boot menu on HDMI, but not on serial.
To be clear, I use the onboard serial (ttyAMA10).

so I have 2 issues
1) I can't force the boot menu to appear unless I add some incorrect boot devices that cause enough delay for the boot menu to show up . Then I can push space on USB keyboard and select 1 (sdcard) instead of 6 (nvme).   This works, but there does not seem to be a documented away to force the boot menu to show up without adding a non bootable device in the boot order

2) the bigger problem is I want this to show up on serial port, not on HDMI.  I'm told this is WAI, but obviously I would really want to have a boot menu over serial to rescue a Pi5 remotely if the main boot device is broken.   I can reboot with sysrq over serial but then I must have a way to force the boot back to sdcard (instead of nvme) so that I can log in and fix the nvme filesystem

Below is what I get with onboard serial (ttyAMA10). I see all the correct output, but it's obviously missing the boot menu that appears on HDMI, so I cannot select another boot media remotely.
```
4.23 RPi: BOOTLOADER release VERSION:2bb2ae64 DATE: 2025/03/10 TIME: 17:10:37
4.30 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1741626637 serial 2aef62b3 boardrev e04171 stc 4230297
4.40 Unsupported boot order 5
4.60 PCIEx1: PWR 1 DET_WAKE 1
4.67 usb_pd_init status 3
4.67 USB_PD CONFIG 0 43
4.76 XHCI-STOP
4.76 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
4.81 USBSTS 1
4.94 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
4.97 xHC0 ports 3 slots 64 intrs 4
4.09 XHCI-STOP
4.09 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
4.14 USBSTS 11
4.16 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
4.21 xHC1 ports 3 slots 64 intrs 4
4.29 USB-PD: src-cap PDO object1 0x0a0191f4
4.30 Current 5000 mA
4.32 Voltage 5000 mV
4.34 Boot mode: USB-MSD (04) order f16
4.38 USB2[2] 00020ae1 connected
4.91 USB2[2] 00200a03 connected enabled
4.92 USB2 root HUB port 2 init
4.14 DEV [01:00] 1.16 000000:02 class 0 VID 046d PID c31c
4.28 HID [01:00] 1.16 000000:02 register HID
4.56 XHCI-STOP
4.56 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
4.60 USBSTS 0
4.64 XHCI-STOP
4.64 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
4.69 USBSTS 18
4.76 XHCI-STOP
4.76 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
4.81 USBSTS 1
4.83 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
4.88 xHC0 ports 3 slots 64 intrs 4
4.00 XHCI-STOP
4.00 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
4.05 USBSTS 19
4.07 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
4.12 xHC1 ports 3 slots 64 intrs 4
4.20 USB2[2] 00020ae1 connected
4.70 USB2[2] 00200a03 connected enabled
4.71 USB2 root HUB port 2 init
4.93 DEV [01:00] 1.16 000000:02 class 0 VID 046d PID c31c
5.03 HID [01:00] 1.16 000000:02 register HID
```

lurch, 2025-07-07:
> 1. I can't force the boot menu to appear unless I add some incorrect boot devices that cause enough delay for the boot menu to show up . Then I can push space on USB keyboard and select 1 (sdcard) [...]

I'm not at the office right now, so can't test, but you might be able to increase the `NET_INSTALL_KEYBOARD_WAIT` or `USB_MSD_STARTUP_DELAY`or `USB_MSD_DISCOVER_TIMEOUT` values to cause the boot menu to appear on-screen? I believe the official recommendation is to hold down space on the USB keyboard _before_ powering on the Raspberry Pi?

It's not quite the full serial-console-boot-menu that you're asking for, but you **may** find the "boot variables" useful https://www.raspberrypi.com/documentation/computers/config_txt.html#the-expression-filter
See also https://www.raspberrypi.com/documentation/computers/config_txt.html#set_reboot_order and https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2712/release-notes.md#2025-01-14-add-set_reboot_order-api-latest

marcmerlin, 2025-07-07:
@lurch  thanks for the variable suggestions. I think careful use of them can allow booting from a partition once, and if it fails reboot from the previous device, basically the A/B boot thing and failsafe boot, but honestly the main thing I'm looking for is something that went bad on the filesystem/device, reboot that may have happened out of my control (including power loss that could cause filesystem corruption and an unbootable system).
At that point, I need at boot time to the system to boot from another device (here sdcard).
Maybe, and I mean maybe, a hack where reboot sets tryboot to the desired nvme partition I want to boot from every time, and if reboot happens due to some other error or sysrq-b or whatnot, tryboot won't be set and then it boots from sdcard.  This almost works except a power failure will cause booting from the wrong partition unattended, so that's not really ideal.
Your other suggestion of set_reboot_order is similar in nature, it is a good thing for purposeful testing, but not really for recovery when things went wrong.
ideally adding a simple boot menu on serial should not be a lot of work, and would bring rPi5 almost on the level of linux with lilo  (editing boot command line like in grub is better, but I can get around that by emergency booting on sdcard, mounting the nvme filesystem, adjusting the command line, and trying again)

timg236, 2025-07-08:
Given the availability of the existing boot menu, tryboot, set-reboot-order, inserting different boot-media or custom initramfs I don't think there is a strong justification for adding another interface to the bootloader which would increase maintenance / testing burden. Closing as not planned.

timg236, 2025-07-08:
Agreed, the bootloaders in rpi-update have hardware watchdog support from power-on to kernel start then kernel start to systemd running.
High numbers in th eAON 'partition' register can be used to detect this and boot a rescue partition / cmdline.
It's currently in rpi-update for regression testing but more docs / examples and OTA features will follow.

lurch, 2025-07-08:
> as long as there is a way to get back to a rescue partition when you get unplanned crashes

@marcmerlin I've got no idea if it supports a UART menu or not, but perhaps you might also like to look at the way that https://github.com/procount/pinn/ works?

marcmerlin, 2025-07-09:
@lurch  PInn or any application can do installs and any interaction over serial port, but booting from an alternate boot media can only be done from the early Pi firmware boot.
Thankfully I know better than to use sdcard which we know will fail sooner or later, and that NVME is a lot more reliable, but long term, NVME also fails, which is why I have 2 in RAID1, plus a backup rescue sdcard that is only used as a rescue filesystem to boot from if NVME goes bad (depending on how it goes bad, if the first stick just stops existing, Pi5 will see the 2nd and boot that instead, but if it's still there on the bus, but not able to respond, then you do need some rescue/recovery method to force the boot to another device at reboot time)

marcmerlin, 2025-07-09:
@timg236  I much prefer an honest answer, even if it doesn't go in the direction I was hoping for than a false promise or expectation :) so thanks for taking the time to answer and being honest in expectations, I appreciate it.

@lurch  of course, I can work around this with a KVM, but the entire point of a serial console is not to have a KVM, which by itself costs many times the price of the Pi5 itself :)    Honestly KVMs are so expensive that getting a full server with built in BMC (basically a 2nd built in computer to do serial and IP) is often cheaper than KVM ports. Also, KVMs require something like 100 to 1000x more bandwidth than a serial connection.
For now this just means a Pi5 is not an appropriate choice for someone who needs a cheap remote server that can be easily controlled remotely (and by that I mean serial port, the solution we've had for about 30 years now, so it's cheap and plentiful), and that's ok.
There is now an RFE for this, if plenty of people end up wanting this, they can vote it up, and if it's just me, then it confirms that it's clearly not a priority. I was told by several people that I shouldn't use an Rpi5 as a small datacenter server, I think they are wrong, but it's true that for someone who needs BMC, as of right now it's not the best choice.

lurch, 2025-07-10:
> if plenty of people end up wanting this, they can vote it up, and if it's just me, then it confirms that it's clearly not a priority.

AFAIK, you're the first person to ask for this.

> I was told by several people that I shouldn't use an Rpi5 as a small datacenter server, I think they are wrong

Seems to work well enough for https://www.mythic-beasts.com/order/rpi 🙂  But I believe they network-boot, rather than booting from an SD card or NVMe drive.

lurch, 2025-07-11:
> One question though for both you and @timg236 : I have 2 NVMEs in my system, they are both detected by the firmware and linux.
Is there any way at all to tell the boot firmware to boot from the 2nd NVME, or it can only boot the first device of each time (sata, usb storage and NVME).

Just to clarify - I don't work on the rpi-eeprom code myself, I just try to help out with answering GitHub issues. So I've got no idea about the answer to your questions, but I'm sure that Tim will know.

marcmerlin, 2025-07-18:
@timg236  thanks for your help and answers earlier. This is what I came up with given my needs and the current boot firmware limitations. Took a while to come up with it, test, and document, but it's done: http://marc.merlins.org/perso/linux/post_2025-07-07_Using-a-Raspberry-Pi-5-_Rpi5_-as-a-Server-With-Btrfs_-Raid1_-Serial-Console-and-Dual-NVME-SD-Card-Recovery-Boot.html
One thing that was an issue is this apparent bug I found where a GPIO goes down to half voltage when you read it (documented in the page above).
I will happily take feedback/suggestions on that page as well as happy to post it in places where it will be useful to others after it's gotten a bit of review/comments/corrections

marcmerlin, 2025-07-18:
And this is known to you I'm sure, but probably not enough other folks, this is what you can now do with SATA and NVME on Pi5, which in my eyes, make them pretty full fledged, servers

Using Raspberry Pi5 as a Server With Raid1, Btrfs, and Multiple NVME M2 or Sata Drives:
http://marc.merlins.org/perso/linux/post_2025-07-07_Using-Raspberry-Pi5-as-a-Server-With-Raid1_-Btrfs_-and-Multiple-NVME-M2-or-Sata-Drives.html

lurch, 2025-07-19:
> I will happily take feedback/suggestions on that page as well as happy to post it in places where it will be useful to others after it's gotten a bit of review/comments/corrections

"if you forgive the somewhat unoptanium 2 pin plug they felt compelled to use"

* It's a 3-pin plug, not a 2-pin
* It's documented at https://datasheets.raspberrypi.com/debug/debug-connector-specification.pdf and https://datasheets.raspberrypi.com/debug/debug-uart-cable-specification.pdf
* A suitable cable is provided with https://www.raspberrypi.com/products/debug-probe/ (which also functions as a USB->UART adaptor)

"sad they didn't just use normal GPIO pins"

I believe that in one of the beta EEPROM images there's now an option to use the UART on the GPIO header instead of the 3-pin UART connector? (The difference here is that the 3-pin UART connector is connected directly to the BCM2712 SoC, whereas the UART on the GPIO header is connected to the RP1 chip (with the RP1 chip then connected to the 2712 over PCIe)).

"I didn't find any good way to remotely power cycle the Pi via a GPIO"

See https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#add-your-own-power-button and https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#power-button

"your boot options can only have a single linux kernel per device
you cannot change the linux boot command line"

You can work around both of these with:
* https://www.raspberrypi.com/documentation/computers/config_txt.html#conditional-filters
* https://www.raspberrypi.com/documentation/computers/config_txt.html#os_prefix
* https://www.raspberrypi.com/documentation/computers/config_txt.html#kernel and https://www.raspberrypi.com/documentation/computers/config_txt.html#cmdline

I dunno what might be going on with your GPIO-voltage issue, but note that you can use https://github.com/raspberrypi/utils/tree/master/pinctrl for more low-level control / diagnosis than using gpiod.

"The first thing to know about initramfs is that it works, but it's not supported by dietpi and likely not by the official distribution either."

I've never used DietPi, but yes Raspberry Pi OS uses initramfs.

lurch, 2025-07-19:
> Using Raspberry Pi5 as a Server With Raid1, Btrfs, and Multiple NVME M2 or Sata Drives:

See also some of the stuff Jeff Geerling has done, if you've not already seen it https://www.youtube.com/watch?v=l30sADfDiM8

marcmerlin, 2025-07-20:
@lurch  thank you kindly for the review and detailed feedback.

Initramfs, I added: The first thing to know about initramfs is that it works, but it's not supported by dietpi and likely not fully by the official distribution either, by that I mean that installing new kernels would automatically make an initramfs, copy it in the right place, and make sure it gets loaded along with that new kernel.
It is supported by the bootloader of course, but what I saw is that installing a new kernel does not automatically do the right thing like it does on my PC linux machines. Is that accurate?

2 pin serial, oops, 3 pins of course, fixed.  The plug, yes, it does exist, and yes you can buy cables that do have it, but my point is that I have literally boxes of cable and connectors of all sizes that I have gathered over 10+ years, and I barely found one that fit the Pi.  It's too late now, but if you look at the POE pins on the board, they use normal sized GPIO  style pins (not the GPIO header, but the same pins and spacing), that's what I meant for serial.  But yes, I bought the new cables with the new special plugs just for it.

Power button: I did read the hardware doc you pointed to and the pins you can solder to, but from what I understood, it was meant to be used with a relay or actual switch.  Can this be wired to the GPIO pin of another Pi? (in theory it might work, but I can tell it was not meant to be used that way)

Changing command line at runtime: I know I can make more conditionals, but my point is that if I suddenly have an unbootable system and needs to add init=/bin/bash or break=premount to the command line, I cannot do this (like you would with grub) unless I have a 2nd full OS I can boot from to edit the vfat partition with cmdline.txt and reboot a 2nd time. This is what I do for now, is there a better way?

I just updated http://marc.merlins.org/perso/linux/post_2025-07-07_Using-a-Raspberry-Pi-5-_Rpi5_-as-a-Server-With-Btrfs_-Raid1_-Serial-Console-and-Dual-NVME-SD-Card-Recovery-Boot.html with your feedback, thank you.

marcmerlin, 2025-07-20:
> > Using Raspberry Pi5 as a Server With Raid1, Btrfs, and Multiple NVME M2 or Sata Drives:
> 
> See also some of the stuff Jeff Geerling has done, if you've not already seen it https://www.youtube.com/watch?v=l30sADfDiM8

Yes, I'm not surprised, my personal testing showed the same, which is why I am working hard on making the Pi5 work despite the few limitations on remote control. Like that other person said, it's now a very capable board.

lurch, 2025-07-20:
> [@lurch](https://github.com/lurch) thank you kindly for the review and detailed feedback.

👍 

> Initramfs, I added: The first thing to know about initramfs is that it works, but it's not supported by dietpi and likely not fully by the official distribution either, by that I mean that [...]

I don't know, that's a question for @XECDesign . But I do know that when `apt upgrade` installs an updated kernel on Raspberry Pi OS, that it automatically builds new initramfs images (one for each kernel arch that you have installed).

> 2 pin serial, oops, 3 pins of course, fixed. The plug, yes, it does exist, and yes you can buy cables that do have it, but my point is that I have literally boxes of cable and connectors of all [...]

🤷  Pi 5 adds a bunch of new connectors - a 2-pin connector for the RTC battery, a 3-pin connector for the UART, and a 4-pin connector for the FAN header. And these are deliberately all different in order to prevent "newbie users" from plugging the wrong cable into the wrong connector. And of course there's the custom PCIe FFC connector as well 😜 

> Power button: I did read the hardware doc you pointed to and the pins you can solder to, but from what I understood, it was meant to be used with a relay or actual switch. Can this be wired to the GPIO pin of another Pi? (in theory it might work, but I can tell it was not meant to be used that way)

I dunno, my low-level hardware knowledge doesn't extend that far. But there are plenty of relay add-on boards for the Raspberry Pi.

> Changing command line at runtime: I know I can make more conditionals, but my point is that if I suddenly have an unbootable system and needs to add init=/bin/bash or break=premount to the command [...]

It's not something that I've used myself, but _maybe_ using a secondary bootloader like U-Boot will allow you to dynamically alter the kernel command line?

marcmerlin, 2025-07-20:
Thanks for the confirmation, I'll then put an update that maybe I hit a dietpi bug, although I did read elsewhere that initramfs does work on rPi5 (indeed it does) but was also "at your own risk" and not supported for upgrades or generally tested.  It's not a huge deal, I just wanted to document it so that users who do rely on me, know to watch out and how to recover if things fail.  If somehow it just works, then they can ignore what I wrote :)

The types of connectors and not confusing what to plug where, fair enough point I suppose.  I'd still have used GPIO like pins for serial since it is GPIO level signalling and nothing bad happens if you short them, but your point is valid.  Either way, I bought some of those custom connectors now, I was just annoyed that they look soo much like the ones I have from ardupilot pixhawk, which has its own 2 to 5 pins tiny ribbon cable and connectors, but they are incompatible by a portion of 1mm :-/

Relay: yes, with an external relay I can emulate a switch.  I was trying to keep things simple, but you are correct that an external relay will work.  Just more wiring and external things dangling out :)

Uboot: funny you mention it, a hardware engineer friend of mine asked me why rPi didn't just use uboot since it does all those things already and has been around for many years.  I have no idea indeed.  I didn't think about getting rPi5 to chain boot uboot and in turn to get that to boot linux (a bit like you can have lilo boot grub, not that you'd want that but you can).  Has anyone actually done it and verified it works, or just a random idea we're discussing?  I'm happy to add a pointer to a uboot solution if one has been done already

lurch, 2025-07-20:
> Relay: yes, with an external relay I can emulate a switch. I was trying to keep things simple, but you are correct that an external relay will work. Just more wiring and external things dangling out :)

If you want a neater power-control solution, there are things like https://www.uugear.com/product/witty-pi-4/ (as well as various UPS add-ons, etc.).

> Uboot: Has anyone actually done it and verified it works

Yes, there are a couple of 3rd-party OSes for Raspberry Pi which use U-Boot, but off the top of my head I can't remember what they are. https://www.google.com/search?q=u-boot+raspberry+pi+5 might point you in the right direction.

peterharperuk, 2025-07-21:
> Is there any way at all to tell the boot firmware to boot from the 2nd NVME

Try adding the following to eeprom config...

```
NVME_CONTROLLER=1
```

timg236, 2025-07-22:
Please can we move this to the forums. This has strayed a long way for the original feature request.

marcmerlin, 2025-07-26:
@timg236  you are correct, apologies. Followups in 
https://forums.raspberrypi.com/viewtopic.php?p=2327154#p2327154
If more people do care about a boot menu over serial, would you like them to vote on this bug or add some kind of +1?  (so that we can track the interest or lack thereof)