raspberrypi/rpi-eeprom #732: [CM4] Exception: No space available - image past EOF

srcshelton, 2025-07-29:
### Describe the bug

```
$ sudo rpi-eeprom-update -a
*** PREPARING EEPROM UPDATES ***

BOOTLOADER: update available
   CURRENT: Fri 16 May 10:34:04 UTC 2025 (1747391644)
    LATEST: Thu 17 Jul 08:26:45 UTC 2025 (1752740805)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader/latest)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 
    LATEST: 
   CURRENT: Fri 16 May 10:34:04 UTC 2025 (1747391644)
    UPDATE: Thu 17 Jul 08:26:45 UTC 2025 (1752740805)
    BOOTFS: /boot
Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.13/rpi-eeprom-config", line 605, in <module>
    main()
    ~~~~^^
  File "/usr/lib/python-exec/python3.13/rpi-eeprom-config", line 585, in main
    image.update_file(args.config, BOOTCONF_TXT)
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python-exec/python3.13/rpi-eeprom-config", line 407, in update_file
    self.update(src_bytes, dst_filename, bootcode)
    ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python-exec/python3.13/rpi-eeprom-config", line 349, in update
    raise Exception('No space available - image past EOF.')
Exception: No space available - image past EOF.
```

The `1747391644` update (and all prior firmware) have installed correctly without error by the same mechanism.

### Steps to reproduce the behaviour

Run `rpi-eeprom-update -a` as the root user.

### Device (s)

Raspberry Pi CM4

### Bootloader configuration.

```
$ sudo rpi-eeprom-config
[all]
#BOOT_UART=0
BOOT_UART=1

# N.B. Not relevant on [pi5], [rpi500], [cm5].
# Set to 2 to use an external GPIO power button.
#
#WAKE_ON_GPIO=1
WAKE_ON_GPIO=0

# N.B. Defaults to 1 on [pi400], [pi500], [cm5].
#
#POWER_OFF_ON_HALT=0

# Boot Order Codes, from [1].
#
# Try SD first (1), followed by NVMe PCIe ([cm4] & [pi5] only), USB
# mass-storage, then loop.
#
#  0x0 - Deprecated
#  0x1 - SD card (or eMMC)
#  0x2 - Network
#  0x3 - RPiBOOT ([cm4])
#  0x4 - USB-MSD
#  0x5 - BCM-USB-MSD (USB2.0 boot from USB-C socket or CM4IO USB-A), not [pi5]
#  0x6 - NVME ([cm4], [pi5])
#  0x7 - HTTP
#  0xe - STOP (display error pattern, reboot needed to continue)
#  0xf - RESTART
#
# [1]: https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
#
#BOOT_ORDER=0xf41

# Set to -1 for infinite retries.
#
#MAX_RESTARTS=-1
#SD_BOOT_MAX_RETRIES=0
#NET_BOOT_MAX_RETRIES=0
#DHCP_TIMEOUT=45000
#DHCP_REQ_TIMEOUT=4000
#TFTP_FILE_TIMEOUT=30000
#TFTP_IP=192.168.1.99
#TFTP_PREFIX=0
#TFTP_PREFIX_STR=
#PXE_OPTION43=Raspberry Pi Boot
#DHCP_OPTION97=0x34695052
#MAC_ADDRESS=
#MAC_ADDRESS_OTP=

# Static IP address
#
#CLIENT_IP=
#SUBNET=
#GATEWAY=

#DISABLE_HDMI=0
#HDMI_DELAY=0

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
#
#ENABLE_SELF_UPDATE=1

#FREEZE_VERSION=0

#HTTP_HOST=fw-download-alias1.raspberrypi.com
#HTTP_PORT=443
#HTTP_PATH=net_install
#IMAGER_REPO_URL=http://downloads.raspberrypi.org/os_list_imagingutility_v3.json
#NET_INSTALL_ENABLED=0
#NET_INSTALL_AT_POWER_ON=0
#NET_INSTALL_KEYBOARD_WAIT=900

# Up to 32 characters, e.g. '6665@169.254.1.1/,6666@/'
# Recommended to activate only by GPIO filter, e.g. '[gpio7=0]'
#
#NETCONSOLE=

# Could be used to boot a recovery partition if the user presses a button via
# GPIO filter, e.g. '[gpio7=0]'
#
#PARTITION=0

#USB_MSD_EXCLUDE_VID_PID=
#USB_MSD_DISCOVER_TIMEOUT=20000
#USB_MSD_LUN_TIMEOUT=2000
#USB_MSD_STARTUP_DELAY=0

#XHCI_DEBUG=0x0

[pi4]
#NET_INSTALL_ENABLED=1
#USB_MSD_PWR_OFF_TIME=1000
#SDRAM_BANKLOW=3
[cm4]
#NET_INSTALL_ENABLED=0
#VL805=0
[cm4s]
#NET_INSTALL_ENABLED=0
[pi5]
#UART_BAUD=115200
#NET_INSTALL_ENABLED=1
#PSU_MAX_CURRENT=
#SDRAM_BANKLOW=1
[cm5]
#NET_INSTALL_ENABLED=0
[none]

# vi: set syntax=cfg:
```

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4

$ sudo vcgencmd bootloader_version
2025/05/16 11:34:04
version 91d7b7b2f21f5434899c3b067951f5afd50a7185 (release)
timestamp 1747391644
update-time 1749695654
capabilities 0x0000007f

$ sudo vcgencmd version
Apr 30 2025 13:33:52 
Copyright (c) 2012 Broadcom
version 5560078dcc8591a00f57b9068d13e5544aeef3aa (clean) (release) (start_cd)

$ uname -a
Linux <hostname> 6.12.25-v8+ #1875 SMP PREEMPT Wed Apr 30 12:32:50 BST 2025 aarch64 GNU/Linux
```

N.B. `/etc/rpi-issue` is potentially misleading - this is a custom Gentoo-based system.

### Bootloader logs

(Please let me know if these could be helpful and I will add details)

### USB boot

I don't think this is related, but briefly:

```
$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 152d:0583 JMicron Technology Corp. / JMicron USA Technology Corp. JMS583Gen 2 to PCIe Gen3x2 Bridge
```

### NVMe boot

n/a

### Network (TFTP boot)

n/a

srcshelton, 2025-07-29:
I reduced the boot configuration down to 516 characters:

```
[all]
BOOT_UART=1
WAKE_ON_GPIO=0
DISABLE_HDMI=1
#FREEZE_VERSION=0
#XHCI_DEBUG=0x0

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
#ENABLE_SELF_UPDATE=1

# Up to 32 characters, e.g. '6665@169.254.1.1/,6666@/'
# Recommended to activate only by GPIO filter, e.g. '[gpio7=0]'
#NETCONSOLE=

# Could be used to boot a recovery partition if the user presses a button via
# GPIO filter, e.g. '[gpio7=0]'
#PARTITION=0

# vi: set syntax=cfg:
```

… which works.

If the firmware image plus boot configuration is too large (and it seems as if there's now only <~600 characters available for boot configuration on CM4?), then could a clearer error message be produced as well as an indication of how much more space is needed/how many characters should be trimmed from the boot configuration in order to succeed?

timg236, 2025-07-29:
The last sector of the EEPROM is reserved for modifiable data, so the EEPROM config fits between the last section and the reserved block. It should be about 1KB now unless some space can be reclaimed elsewhere.
The Python no space Exception error handler could probably be improved.

timg236, 2025-07-29:
Yeah, that sounds right. Looks like the Pi4 code size can be squeezed a bit reverting the optimize-for-speed that was added when the Pi5 code was merged (will leave it on for Pi5)