raspberrypi/rpi-eeprom #734: RPi 5 boot get s stuck on "fs_open: 'armstub8-271.bin'" when USB peripheral is plugged-in

jrodonnell, 2025-08-05:
### Describe the bug

I have an RPi 5 running Raspberry Pi OS Lite 64 bit off an SD card and I have a consistent issue with booting while a USB peripheral is plugged-in, in my case a USB keyboard. If the peripheral is plugged-in during boot, the boot process will hang at the message `fs_open: 'armstub8-271.bin'` and basically stay there forever. If I unplug the USB device and reboot, everything comes up normally. I have not made any modifications to the boot config or Raspberry Pi OS image, this is just by using the default latest image available via Imager v1.8.5.

### Steps to reproduce the behaviour

On a Raspberry Pi 5 with Raspberry Pi OS Lite 64 bit installed:

1. Plug in a USB peripheral (e.g. keyboard)
2. Reboot
3. Boot will hang at `fs_open: 'armstub8-271.bin' `
4. Unplug USB peripheral
5. Reboot
6. OS boots properly to shell login screen

### Device (s)

Raspberry Pi 5

### Bootloader configuration.

```
$ rpi-eeprom-config
[all]
BOOT_UART=1
BOOT_ORDER=0xf461
NET_INSTALL_AT_POWER_ON=1
```

timg236, 2025-08-05:
Please add full details of the peripheral, otherwise, it's not possible to reproduce the issue.

jrodonnell, 2025-08-12:
Here are some details about my RPi setup:
- Raspberry Pi OS Lite 64 bit (2025-05-13) burned to NVMe drive via Imager 1.8.5
- Official 27W power supply
- Official fan kit
- 256 GB NVMe drive on HAT
- USB keyboard (no hub)
- MicroHDMI to a Gigabyte monitor

This problem happened during the first boot and does not occur anymore. I rebooted several times and got stuck in the same spot every time, and only when I unplugged my keyboard did it boot all the way up properly. I don't know if there's some initialization done on the first boot that is getting hung up, but it was consistently getting stuck on that step until I unplugged my nondescript USB keyboard.

Perhaps my diagnosis is wrong or just a coincidence, but I am not the only one who has encountered issues with the RPi5 boot getting stuck on this armstub8-271.bin step. There is a post in the forums with many people raising the issue as well: https://forums.raspberrypi.com/viewtopic.php?t=363184

timg236, 2025-08-12:
I can't see any details of the keyboard which is causing the USB issues.

closed 2025-10-21 by timg236