raspberrypi/rpi-eeprom #751: Pi4 not booting after power cycle with multiple USB disks since eeprom update on Aug 22, 2025

hungaborhorvath, 2025-09-21:
### Describe the bug

I have a Raspberr Pi 4 in an Argon One M.2 case with an M2 SSD in the case, and 3 external drives attached via a powered USB3 HUB. The Pi is powered by the official power supply. There is only one boot partition in the four drives altogether, which is on the M2 drive in the case.

Before the 2025 Aug 22 update to the latest eeprom, the Pi booted properly with this setup. After that update, it can't boot up from a power loss. The led is stable green, not blinking at all. 
If I remove the powered USB3 HUB, the pi4 boots up as it should. Then I can reattach the powered HUB with the 3 external drives after boot and use the system as normal.

I use the Pi4 headless, but for trying to diagnose this, I plugged a monitor to it. I shut it down, then pulled the power cable out then plugged it in, and the monitor did not power up, nor did the Pi4 boot up. Led was solid green. When I removed the USB3 HUB, power cycled it, and the Pi4 booted properly. Interestingly, when I plugged in the USB3 HUB and rebooted the pi4 using the reboot command, it successfully rebooted. Then when shut down and did a full power cycle, again, no boot, not even monitor getting power. Led is solid green. 
Only could boot up if first I unplugged the USB 3 HUB, and then Pi4 booted properly.

I did one more eeprom update to latest on September 9, 2025, but the behaviour is not corrected.

Happy to provide more details as needed, though the Pi4 is not easily accessible physically, but have easy remote access to give outputs of commands, etc.

### Steps to reproduce the behaviour

See in the description above, power cycle does not boot up, gives solid green light, no HDMI output, either. Removing the USB3 HUB and the Pi4 boots up with the M2 SSD in the Argon ONE M2 case. 

No issues before the Aug 22 update for the eeprom.

### Device (s)

Raspberry Pi 4 Mod. B

### Bootloader configuration.

rpi-eeprom-config 
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

### System

cat /etc/rpi-issue
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage5

vcgencmd bootloader_version
2025/08/27 19:57:12
version 000d3ca289ae790a2e46225eb11b319c67642b1a (release)
timestamp 1756321032
update-time 1757372477
capabilities 0x0000007f

vcgencmd version
Apr 30 2025 13:33:39 
Copyright (c) 2012 Broadcom
version 5560078dcc8591a00f57b9068d13e5544aeef3aa (clean) (release) (start)

uname -a
Linux raspberrypi4-db 6.12.34+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux

### Bootloader logs

I may not be tech-savvy to do this UART thing I'm afraid. There was no HDMI diagnostics screen, the monitor didn't even boot up.

### USB boot

Bus 002 Device 003: ID 174c:1156 ASMedia Technology Inc. Forty
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x174c ASMedia Technology Inc.
[... 579 lines cut ...]
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12

### NVMe boot

The drive is not NVME

### Network (TFTP boot)

This should not be applicable.

hungaborhorvath, 2025-11-08:
Well, today I spent time on this and changed the powered USB HUB to a different powered USB HUB, and indeed, the reboot issues went away, based on some initial testing. It is weird, because it looks that the old powered HUB also powered the drives but maybe not with the same wattage, maybe due to wear and tear? Without further tools I won't be able to debug this any further, but I'll close the issue as solved, and likely cause being USB pwered HUB not giving enough power to the USB drives at boot. 

Thank you for the help! I wouldn't have thought of this otherwise.

closed 2025-11-08 by hungaborhorvath

timg236, 2025-11-09:
Thanks for updating the Github issue, we do keep an eye on these things even if there is no resulting software change. If you have details of good/bad USB hubs then it's good to know.